brocaar/chirpstack-network-server #496: Possible regression due to code merge / lost redis SSL support

bschuedzig, 2020-08-26:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?

network-server might have lost redis SSL support due to a merge conflict.

The commit https://github.com/brocaar/chirpstack-network-server/commit/73579259a1f2fb7ec9b449dd6e7e55567f6a900f#diff-5be1943b832fb6e1cd1468273c2d82c4 (Mar 31, 2020) introduced new features. One of them is parsing `REDIS_URL`. This commit is present on both network-server and application-server.

The commit https://github.com/brocaar/chirpstack-network-server/commit/0ef375ccb891d80dbae131efd3e309cc6dc8b9d9#diff-5be1943b832fb6e1cd1468273c2d82c4 (Jul 20, 2020) introduced a possible code regression, effectively removing redis SSL support (by not calling Redis.ParseUrl()) anymore.

## What did you expect?

That the second commit does not remove the previous functionality (which is still present in network-server)

## Steps to reproduce this issue

Steps:

not applicable

## Could you share your log output?

not applicable

## Your Environment

all

brocaar, 2020-08-27:
I think that would be needed is some additional fields to configure the `TLSConfig` field on `redis.ClusterOptions`, `redis.FailoverOptions` and `redis.Options`.

BertKappa, 2020-10-29:
Any update? We tried to upgrade to the last version, but our Redis (DigitalOcean) is not supported because it requires SSL/TLS. We always get the error "msg="gateway/mqtt: acquire lock error" error="acquire lock error: EOF""

brocaar, 2020-10-30:
@BertKappa I believe the above commit should resolve your issue.

If somebody else would like to extend this by adding configuration for CA and client certificates, that would be great :)

closed 2020-10-30 by brocaar

bschuedzig, 2020-10-30:
Awesome, thank you @brocaar !
Do you already have an idea when this could be part of a release? That would allow us to remove a weird workaround from the cluster :)

BertKappa, 2020-11-02:
Thanks @brocaar ! Same question as @bschuedzig . Any idea when this will be released? So we can upgrade using the Ubuntu package manager?

commit 7e5685e2 (2020-10-30): Add tls_enabled option to redis config.

Note: this currently uses InsecureSkipVerify=true. Potentially this can
be extended in the future with additional fields for ca_cert
configuration.

Closes #496.