brocaar/chirpstack-network-server #519: DeviceTimeReq: add extra validity check for gateway time

darkfader, 2021-01-14:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## Summary

Improve "fallback on current server time when time field is not available".
Use fallback also when the gateway time seems too old.

## What is the use-case?

Time field may be wrong (not synced) but still sent to the server. Our gateways boot with 2012-01-01 and are synchronized to the nodes if they happen to join and request time during this period.

## Implementation description

if (timeSinceGPSEpoch == 0   or   gatewaytime < servertime - 10 minutes) { ... use fallback

## Can you implement this by yourself and make a pull request?

no

brocaar, 2021-02-02:
I understand the issue, but I'm not a fan of having to do the validation at every function within the NS. The NS time might be skewed too, in which case a wrong time is sent to the devices as well as when no GW timestamp exists, it will fallback onto the NS server time.

Ideally this should be solved at the packet-forwarder level. I do believe that if no GPS fix is available, the `time` field is left empty. Which is why the Gateway Bridge `fake_rx_time` option exists. If that is turned off then without a GPS fix no `time` is reported and the NS will always use its server time.

Alternatively, would it be possible to force a NTP sync of the gateway before starting the packet-forwarder?

bconway, 2021-02-11:
Past timestamps are also expected if the gateway has lost connectivity and the messages are queued (QoS = 1) by the Gateway Bridge/MQTT client on the gateway. I have written custom logic to support time windows, for example, up to the expected life of a gateway battery.

brocaar, 2021-03-31:
I'm closing this issue as I think this doesn't fix the actual issue. There is no guarantee that the server time is more accurate than the gateway time. The gateway might have a GNSS time-source and the server is not time-synced (e.g. a Pi booting with a 2012-01-01 timestamp). In this case it would drop the accurate timestamp provided by the gateway and fallback to the server time.

closed 2021-03-31 by brocaar