brocaar/chirpstack-network-server #522: Support for LR-FHSS (was LoRa-E)

billversteeg, 2021-02-01:
Are there any plans for supporting LR-HFSS (was LoRa-E)? I know that it is being rolled into the next release of LoRaWAN, but I am not sure if you are putting it into Chirpstack.

By the way, I really like chirpstack, and appreciate the work you are putting into it.

Bill VerSteeg

brocaar, 2021-02-02:
Hi @billversteeg, yes I'm planning to support this when this becomes part of the specs. Do you know of any public available implementations which I could use for testing this end-to-end?

billversteeg, 2021-02-02:
I have a “Technology Adapter Program” agreement with Semtech, and a
commercial project that is probably going to deploy LR-FHSS. We are going
to use Semtech’s NS for the early phases of the project.

I may be able to send you a couple of gateways and some sensors, if that
would help.

Bvs

brocaar, 2021-02-04:
Hi @billversteeg, yes that would be really helpful for testing at my side! Would you mind sending me an email at info@brocaar.com so that I could follow up with my details in private? Thanks!

aschliemann, 2021-06-16:
Hi @brocaar,
having the latest docker chirpstack-server pack (as,ns,etc) I am getting a lot of as-errors complaining LR_FHSS modulation unknown...
Have you managed to managed to update chirpstack regarding this modulation or have I configured something stupidly.....
Thanks.

> cs_as_1     | time="2021-06-16T06:21:32.268740059Z" level=error msg="finished client unary call" ctx_id="<nil>" error="rpc error: code = Unknown desc = unknown modulation: LR_FHSS" grpc.code=Unknown grpc.ctx_id=6cdb298d-271e-4bf9-86dd-30f507ccc549 grpc.duration="659.978µs" grpc.method=SendProprietaryPayload grpc.service=ns.NetworkServerService span.kind=client system=grpc
> 
> cs_as_1     | time="2021-06-16T06:21:32.269089704Z" level=error msg="send gateway ping error: send ping error: send proprietary payload error: rpc error: code = Unknown desc = unknown modulation: LR_FHSS"
> 
> cs_as_1     | time="2021-06-16T06:21:33.273075969Z" level=error msg="finished client unary call" ctx_id="<nil>" error="rpc error: code = Unknown desc = unknown modulation: LR_FHSS" grpc.code=Unknown grpc.ctx_id=5030cd10-dd98-4bd8-a4fc-2033e84d775c grpc.duration="685.9µs" grpc.method=SendProprietaryPayload grpc.service=ns.NetworkServerService span.kind=client system=grpc
> 
> cs_as_1     | time="2021-06-16T06:21:33.273447763Z" level=error msg="send gateway ping error: send ping error: send proprietary payload error: rpc error: code = Unknown desc = unknown modulation: LR_FHSS"
> 
> cs_as_1     | time="2021-06-16T06:21:34.276881712Z" level=error msg="finished client unary call" ctx_id="<nil>" error="rpc error: code = Unknown desc = unknown modulation: LR_FHSS" grpc.code=Unknown grpc.ctx_id=00f3d3f0-34d5-494e-bbf5-2270792411fd grpc.duration="660.984µs" grpc.method=SendProprietaryPayload grpc.service=ns.NetworkServerService span.kind=client system=grpc
> 
> cs_as_1     | time="2021-06-16T06:21:34.27722744Z" level=error msg="send gateway ping error: send ping error: send proprietary payload error: rpc error: code = Unknown desc = unknown modulation: LR_FHSS"
> 
> cs_as_1     | time="2021-06-16T06:21:35.281041979Z" level=error msg="finished client unary call" ctx_id="<nil>" error="rpc error: code = Unknown desc = unknown modulation: LR_FHSS" grpc.code=Unknown grpc.ctx_id=a32b3ffc-4b70-433c-9365-bf7e92b8108d grpc.duration="815.785µs" grpc.method=SendProprietaryPayload grpc.service=ns.NetworkServerService span.kind=client system=grpc
> 
> cs_as_1     | time="2021-06-16T06:21:35.281406092Z" level=error msg="send gateway ping error: send ping error: send proprietary payload error: rpc error: code = Unknown desc = unknown modulation: LR_FHSS"
> 
> cs_as_1     | time="2021-06-16T06:21:36.285115606Z" level=error msg="finished client unary call" ctx_id="<nil>" error="rpc error: code = Unknown desc = unknown modulation: LR_FHSS" grpc.code=Unknown grpc.ctx_id=fdc5471d-b35d-4662-a265-6c0b74352597 grpc.duration="797.175µs" grpc.method=SendProprietaryPayload grpc.service=ns.NetworkServerService span.kind=client system=grpc
> 
> cs_as_1     | time="2021-06-16T06:21:36.285455189Z" level=error msg="send gateway ping error: send ping error: send proprietary payload error: rpc error: code = Unknown desc = unknown modulation: LR_FHSS"

brocaar, 2021-06-16:
I'm working on it, stay tuned :)

xvilajosana, 2021-06-16:
awesome!!

brocaar, 2021-09-07:
This has already been integrated in the latest version :)

closed 2021-09-07 by brocaar