brocaar/chirpstack-network-server #526: Include the gateway's name on gateway stats messages

JohnRoesler, 2021-03-25:
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## Summary

It would be helpful if the gateway stats message structure included the gateway's name as it appears in chirpstack

## What is the use-case?

I consume the gateway stats messages off of the mqtt topic by which they're sent up from the gateway bridge. The gateway id is not the primary key in my other system of record for gateways. I am thinking I could set the name of the gateway to be that primary key and therefore be able to process the gateway stats messages into my system without an additional lookup using the gateway ID.

## Implementation description

I imagine this would mean adding gateway name to https://github.com/brocaar/chirpstack-api/blob/master/go/as/as.pb.go#L628-L650. However, I am not sure if the gateway bridge has access to the gateway name.

Another possible option I could see, is adding a new application server output to mqtt that provides the gateway stats messages in json with additional detail known by the application server (gateway name, tags, etc.)

## Can you implement this by yourself and make a pull request?

yes - once the design is known / confirmed

brocaar, 2021-03-29:
Hi @JohnRoesler, the ChirpStack Gateway Bridge has no access to the gateway name. Also I'm not really in favor of letting the ChirpStack Application Server re-expose the gateway stats with the name added.

In this specific case it might be better to have a small services consuming the `gateway/ID/event/stats`, which will then retrieve the name of that gateway and re-publish the message to a different topic (or ingest it directly in your other system). Potentially you could use the same protobuf message and store the name in the `meta_data` key/value field :)

JohnRoesler, 2021-03-29:
@brocaar I figured as much 😄  i built an a component that looks up and caches the gateway details

closed 2021-03-29 by JohnRoesler