brocaar/chirpstack-network-server #539: Downlink MIC calculation is incorrect when Acknowledging a confirmed uplink in LoRaWAN 1.1

jamesl-dm, 2021-07-02:
- [x ] The issue is present in the latest release.
- [x ] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
In LoRaWAN 1.1, the downlink MIC uses a block called B0, detailed in section 4.4.1 of the standard.
The block contains a field ConfFCnt, which is usually filled with zero.
When the downlink acknowledges a confirmed uplink, it is filled with the frame count of that uplink, mod 65536.
However, Chirpstack 3.13.0 seems to fill in the uplink frame count, **plus 1**.

## What did you expect?
The ConfFCnt field should use the uplink frame count.

## Steps to reproduce this issue
Steps:
1. Register an OTAA device set for LoRaWAN 1.1
2. Send a confirmed uplink from the device
3. The replying downlink will fail its MIC check, because the frame count is off by one in the B0 block

## Could you share your log output?
N/A

## Your Environment
Sorry - I'm not sure how to get this information out of the Docker environment.
But the network server seems to be 3.13.0, and I think this bug has been in there for years.

abalmos, 2021-07-03:
I can confirm this behavior. One of our v1.1 devices works fine under this bug because it only uses unconfirmed uplinks; however, our other (spec compliant) device uses confirmed uplinks (when sending a RekeyInd commands I believe), essentially rendering the connection lame (no adr, frequent rejoins, etc.).

abalmos, 2021-07-07:
I've tested #540 with the device inspiring this issue. Maybe Digital Matter can confirm the fix @jamesl-dm ?

jamesl-dm, 2021-07-07:
Sorry - my schedule is a bit tight right now... Not even sure _how_ to build chirpstack from source.

PR #540 (2021-07-07): Correctly compute ConfFCnt for confirmed uplink's downlink MIC
Currently, `ConfFCnt` in the MIC is always one larger than the `FCnt` of the corresponding uplink, as described in #539. This is because `ctx.DeviceSession.FCntUp` was already incremented by the initial processing of the uplink [here](https://github.com/brocaar/chirpstack-network-server/blob/4b18f02316b5a9990847e17cc320fb69ce3fc589/internal/uplink/data/data.go#L60).

I can confirm this PR enables a LoRaWAN v1.1 device to successfully complete ADR.

I did not add a test because I am still trying to understand the framework--this is  [... cut]
commit 82c3d7f4 (2021-07-07): Correctly compute ConfFCnt for MIC. (#540)

Currently `ConfFCnt` in the MIC is always one larger then the `FCnt` of the corresponding uplink, as described in #539. This is because `ctx.DeviceSession.FCntUp` was already incremented by the initial processing of the uplink.

Closes #539.
closed 2021-07-07 by brocaar