brocaar/chirpstack-network-server #542: rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetDeviceActivation

finyyu, 2021-07-16:
- [ √ ] The issue is present in the latest release.
- [ √ ] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
when saved device's [keys(OTAA)],  network server log shows "error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetDeviceActivation". After this operation, the device cannot join.

## What did you expect?
The device can join success, and can send request to lora server.

## Steps to reproduce this issue

Steps:

1. Add network-servers, gateway-profiles, service-profiles and device-profiles(supports OTAA)
2. Add gateway, application and device, set [KEYS(OTAA)]
3. Look network-server log, show [error message], details followed in next sectioni
4. Using device to join,  there is only [JoinRequest], but not response of [JoinAccept].

## Could you share your log output?
1、Set [KEYS(OTAA)]
[network-server] logs
7月 16 11:37:11 raspberrypi chirpstack-network-server[22258]: time="2021-07-16T11:37:11.330377246+08:00" level=info msg="finished unary call with code NotFound" ctx_id=9fd76f48-0c24-4dab-9eab-b417d7e8874e error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetDeviceActivation grpc.service=ns.NetworkServerService grpc.start_time="2021-07-16T11:37:11+08:00" grpc.time_ms=0.903 peer.address="127.0.0.1:42286" span.kind=server system=grpc

[application-server] logs
7月 16 11:37:11 raspberrypi chirpstack-application-server[22407]: time="2021-07-16T11:37:11.205210584+08:00" level=info msg="device-keys updated" ctx_id=8c5547df-2482-4f0e-b3e7-f5124877c1f2 dev_eui=85c70da1adf86e43
7月 16 11:37:11 raspberrypi chirpstack-application-server[22407]: time="2021-07-16T11:37:11.207132492+08:00" level=info msg="finished unary call with code OK" ctx_id=8c5547df-2482-4f0e-b3e7-f5124877c1f2 grpc.code=OK grpc.method=UpdateKeys grpc.service=api.DeviceService grpc.start_time="2021-07-16T11:37:08+08:00" grpc.time_ms=2673.328 peer.address="[::1]:49708" span.kind=server system=grpc
7月 16 11:37:11 raspberrypi chirpstack-application-server[22407]: time="2021-07-16T11:37:11.332567558+08:00" level=info msg="finished client unary call" ctx_id=204885d6-5ee4-4c02-97bb-6b2912d097fc error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.ctx_id=9fd76f48-0c24-4dab-9eab-b417d7e8874e grpc.duration=3.67365ms grpc.method=GetDeviceActivation grpc.service=ns.NetworkServerService span.kind=client system=grpc
7月 16 11:37:11 raspberrypi chirpstack-application-server[22407]: time="2021-07-16T11:37:11.333283589+08:00" level=info msg="finished unary call with code NotFound" ctx_id=204885d6-5ee4-4c02-97bb-6b2912d097fc error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetActivation grpc.service=api.DeviceService grpc.start_time="2021-07-16T11:37:11+08:00" grpc.time_ms=11.999 peer.address="[::1]:49708" span.kind=server system=grpc

2、join
[network-server] logs
7月 16 11:40:13 raspberrypi chirpstack-network-server[22258]: time="2021-07-16T11:40:13.611008329+08:00" level=info msg="gateway/mqtt: uplink frame received" gateway_id=d35a141f592e347d uplink_id=00000000-0000-0000-0000-000000000000
7月 16 11:40:13 raspberrypi chirpstack-network-server[22258]: time="2021-07-16T11:40:13.814983561+08:00" level=error msg="uplink: processing uplink frame error" ctx_id=66cd84b4-d9af-4436-92f9-9d68382273a7 error="zero items in collect set"

[application-server] logs 
no

```shell
journalctl -f -n 100 -u chirpstack-network-server
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | 3.15.0  |
| Network Server      | 3.13.0  |
| Gateway Bridge      |  3.11.0 |
| Chirpstack API      |     |
| Geolocation         |    no     |
| Concentratord       |         |

finyyu, 2021-07-16:
OS infomation：
Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

brocaar, 2021-07-19:
> when saved device's [keys(OTAA)], network server log shows "error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetDeviceActivation".

That is expected as after saving the root keys, the device still needs to activate first to derive the session-keys, so it is correct that no activation exist yet.

> raspberrypi chirpstack-network-server[22258]: time="2021-07-16T11:40:13.814983561+08:00" level=error msg="uplink: processing uplink frame error" ctx_id=66cd84b4-d9af-4436-92f9-9d68382273a7 error="zero items in collect set"

You have to debug the "zero items in collect set". It means that after the de-duplication timeout, there is no data collected. My assumption is a misconfiguration, causing the data to expire too soon.

As I think this is a configuration / setup issue (I can not reproduce it), I would like to ask you to post your question in https://forum.chirpstack.io.

closed 2021-07-19 by brocaar