brocaar/chirpstack-network-server #543: class-c downlink queue processing issue. NS sends mac-commands but never reach payload data.

fancar, 2021-07-19:
- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
ns keep constantly sending downlinks to class-c device, which is offline if a message in the device queue for the device is present. It happens because every iteration ns sends mac-commands in frame payload, not application data. So the task won't be removed from the queue during ACK.

## What did you expect?
ns shell not send mac-commands at all.

## Steps to reproduce this issue
Steps:
1. add a new class-c device.
2. Switch it off after it sends one uplink.
3. put a message in a queue.
4. in case if ns will decide to send a >15 bunch commands in the frame payload, you will see the loop:  ns sends downlinks every 2 seconds (lock time).
5. 
## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.13  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

fancar, 2021-07-19:
This description I've found in TS001-1.0.4 LoRaWAN® L2 1.0.4 Specification
> 
> A Class C downlink SHALL NOT transport any MAC command. If an end-device receives a
> Class C downlink containing a MAC command, either in the FOpts field (if FPort is either
> missing or > 0 ) or in the FRMPayload field (if FPort=0 ), it SHALL silently discard the entire
> frame.

If it's stil relevant, then the solution is just remove setMACCommandsSet from scheduleNextQueueItemTasks

brocaar, 2021-08-09:
Thanks @fancar , the above commit fixes the issue.

commit ad1c90a2 (2021-08-09): Remote set mac-command set from scheduleNextQueueItemTasks.

Per LoRaWAN 1.0.4 specs, Class-C downlinks shall not transport any
mac-commands. This does also remove sending mac-commands for Class-B.

Closes #543.
closed 2021-08-09 by brocaar