brocaar/chirpstack-network-server #544: Integration with LoRaCloud Join Server

WisangAnggoro, 2021-07-26:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?

Hi! I want to integrate my Chirpstack Network Server with LoRa Cloud Join Server to connect my LR1110 device with the production keys. I've already changed the server on the chirpstack-network-server.toml section [join_server.default] (from http://localhost:8003 into https://js.loracloud.com:7009), and also add the TLS Credential for the Join Server (LoRaCloud provide the credentials in trust, crt, and key file). However, after LR1110 devices try to join, an error happened on the device data says that "join-server returned error: unmarshal response error: invalid character ':' after top-level value".

## What did you expect?

Device LR1110 (that already registered and claimed on LoRa Cloud) can join OTAA with only DeviceEUI. AppKey is handled by the device and LoRa Cloud Join Server. This job is already done by using ns.loracloud.com, but I can't find how they configure it. 

## Steps to reproduce this issue

Steps:
1. Download TLS Credentials on LoRa Cloud for the Join Server
2. On chirpstack-network-server.toml, set  server="https://js.loracloud.com:7009",  ca_cert, tls_cert, and tls_key using the file provided by step 1
3. restart Chirpstack NS
4. Try join OTAA using LR1110 and check the device data

## Log output

```shell
level=info msg="finished client unary call" ctx_id=e422db30-5cf9-4c9c-8d1c-e3f11901d472 grpc.code=OK grpc.ctx_id=91b3b9e4-4f51-47c7-b052-227195e28d43 grpc.duration=3.638175ms grpc.method=HandleGatewayStats grp>
level=info msg="gateway/mqtt: uplink frame received" gateway_id=60c5a8fffe76f6c9 uplink_id=feafd70e-7237-4519-99fe-ae14734febda
level=info msg="uplink: frame(s) collected" ctx_id=b116c198-e91f-4c9a-8aef-c70585bfa017 mtype=JoinRequest uplink_ids="[feafd70e-7237-4519-99fe-ae14734febda]"
level=info msg="finished client unary call" ctx_id=b116c198-e91f-4c9a-8aef-c70585bfa017 grpc.code=OK grpc.ctx_id=6e8d8f88-8850-49a8-9945-709206853545 grpc.duration=2.436254ms grpc.method=HandleError grpc.servi>
level=error msg="uplink: processing uplink frame error" ctx_id=b116c198-e91f-4c9a-8aef-c70585bfa017 error="join-request to join-server error: unmarshal response error: invalid character ':' after top-level val>
```

## Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.16.0 |
| Network Server      | v3.14.0 |
| Gateway Bridge      | v3.12.0 |

brocaar, 2021-09-07:
I have tested this again and it works fine, **BUT** please note that the **LoRa Cloud Join Server** given in the LoRa Cloud web-interface is not the endpoint which you must configure in the `chirpstack-gateway-bridge.toml` config file.

You will find the actual endpoint in this documentation:
https://www.loracloud.com/documentation/join_service?url=rens_api.html#joinreq-joinans

Example:

```toml
  [[join_server.servers]]
  join_eui="0016c001fffe0001"
  server="https://js.loracloud.com:7009/api/v1/rens/rens-XYZ/lbi"
  ca_cert="/path/to.trust"
  tls_cert="/path/to.crt"
  tls_key="/path/to.key"
```

You will find the Rens ID in the LoRa Cloud web-interface under **Your network servers**.

closed 2021-09-07 by brocaar

brocaar, 2021-09-07:
I've added this documentation: https://www.chirpstack.io/network-server/features/join-server/.