brocaar/chirpstack-network-server #548: redis password not set in sentinel mode

gbrehmer, 2021-09-15:
The go client FailoverOptions have two password fields `SentinelPassword` AND `Password` (https://github.com/go-redis/redis/blob/7ecd5cee5b1b56fd5135b1b38207ae12ee9e70b4/sentinel.go#L49). Last one is used for the client connection. But chirpstack is only setting the sentinel password: https://github.com/brocaar/chirpstack-network-server/blob/8bc17e3a25091debba7be3072b3c1119872d6907/internal/storage/storage.go#L68
For backwards compatibility probably its better to introduce a new property, but in our case we use the same password for sentinel and redis, so it would be enough to only set the additional property.

gbrehmer, 2021-09-15:
ist already done in the application server: https://github.com/brocaar/chirpstack-application-server/blob/efb3284a2f04938180d9bd818f647b86177c6cc1/internal/storage/storage.go#L97 so probably no backwards compatibility is required :)

brocaar, 2021-10-01:
If it is already there in the ChirpStack Application Server code, then it makes sense to add it to the NS code-base as well. Would you like to create a pull-request for this?

closed 2021-10-01 by brocaar

fmendez89, 2021-11-05:
@brocaar Thank you very for the release and the project itself.

PR #550 (2021-10-01): sentinel-mode: missing redis pw
fixes #548