brocaar/chirpstack-network-server #554: Error using AS923_4 with Docker compose install

t0mmyz7, 2021-10-20:
Hi

I am getting the following error trying to start the stack with docker-compose:
`chirpstack-network-server_1      | time="2021-10-20T09:50:54.784674302Z" level=fatal msg="setup band error: get band config error: lorawan/band: band AS923_4 is undefined"`

my network server config is:
```

[postgresql]
dsn="postgres://chirpstack_ns:chirpstack_ns@postgresql/chirpstack_ns?sslmode=disable"

[redis]
url="redis://redis:6379"

[network_server]
net_id="000000"

[network_server.band]
name="AS923_4"

[network_server.gateway.backend.mqtt]
server="tcp://mosquitto:1883"

[join_server.default]
server="http://chirpstack-application-server:8003"

```

I try changing the docker image version to 3.15.1 and 3.15.2 and it didn't help

am I setting it wrong?

IoTThinks, 2021-11-03:
@t0mZ4 After you add the network server into application server, will the application server shows as AS923-4 or EU868?

I added AS923-2 OK. However, the application server shows as EU868.
My version is v3.15.3.
https://forum.chirpstack.io/t/as923-2-shows-as-eu868-on-network-server-v3-15-3/12579
[image]

Thanks a lot.

t0mZ4, 2021-11-03:
hi mate you are correct, mine also shows EU868, i haven't noticed that.

I can confirm that we tested AS923-4 unit and it works, so this might be a display error?

[image]

@brocaar what do you think?

IoTThinks, 2021-11-03:
@t0mZ4 
Ok if it works then it may be a cosmetic issue.

Let me try if my as923-2 works while waiting this to be fixed.

Thanks a lot for your confirmation.

brocaar, 2021-11-09:
Yes, this is a cosmetic issue. I'll fix this in the next (minor) release.

IoTThinks, 2021-11-09:
@brocaar Thanks for your reply.
Ok, if cosmetic issue, then I will go ahead with futher testing.
Thanks a lot.

IoTThinks, 2021-11-16:
Hi all,
AS923-2 is confirmed working with my AS923-2 gateway and device.
This is confirmed a comestic issue only.

Thanks all for your help.

closed 2025-05-02 by brocaar