brocaar/chirpstack-network-server #566: For the same created_at: Duplicate dev_nonce's,

danieroux, 2021-12-15:
- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?

## What did you expect?

## Steps to reproduce this issue

Steps:

1.
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

csanso-limit, 2021-12-17:
This may be a stretch but I think #557 is related to this.
I think I've experienced the same errors as you but with a single gateway, although the problem makes more sense with multiple gateways.

> Duplicate dev_nonce's, for the same dev_addr and created_at leads to data being lost.

If you notice closely the created_at is not exactly the same, it's off by 3767 microseconds.

> so my guess is that the JOIN does not de-duplicated?

I'm pretty sure that is exactly the problem, they are not getting de-duplicated.
The de-duplication logic probably needs a rethink, at least in the case of JOINs.
Because it does seem like JOINs are absent from being de-duplicated.
It could also be that de-duplication does not take different timestamps or frequencies into consideration.

csanso-limit, 2021-12-17:
I am curious to know what dev_addr does the device endup receiving from the network server?
The first JOIN dev_addr, the second JOIN dev_addr or none?

brocaar, 2022-04-28:
This has been fixed by https://github.com/brocaar/chirpstack-network-server/commit/1b505944a63ebdb8b3556211d8f9e76e606f516a.

closed 2022-04-28 by brocaar