brocaar/chirpstack-network-server #567: chirpstack-network-server config file over-rides settings for RXWIN1 in Device Profile

petertoome, 2021-12-20:
- [ Y] The issue is present in the latest release.
- [Y ] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
We are usingthe AU915 channel plan with devices on SB1 and SB2.
We are using ABP mode on private networks
Our suppleir uses Acknowledged packets in order to ensure we get a contiguous data set.
The ABP mode devices stopped processing ACK packets.
When a devices is first enabled it will receive the ACK, but after 1 or 2 packets the device stops receving them

## What did you expect?
The devices send an Uplink packet then wait for an ACK to determine if it has been received. If not, they retry. 
The ACK should be received and the next packet sent.

## Steps to reproduce this issue

Steps:

1. Try multiple devices
2. Try multiple channel plans
3. Try multiple MAC versions
4. Monitor uplink and downlink packets for compliance with Regional parameters

## Could you share your log output?

 Successful packet:

923900000 | timerstamped | 1639605171193 | 3063487003 | 0 | 27 | 8 | true | LORA | SF7BW500 | 4/5 | true | 25 | YLYfDSYNAQACEQEDAAAAcQMAAP8B9OS2DQ== | Unconfirmed                                       Data Down | 0 | 0 | 260D1FB6
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

Failed packet:

927500000 | timerstampe | 1639609839462 | 718444108 | 0 | 27 | 8 | false | LORA | SF12BW500 | 4/5 | true | 18 | YLYfDSYGewACEQIGCQ2+2XC+ | Unconfirmed                                       Data Down | 0 | 0 | 260D1FB6
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

The successful packet is sent with the correct RX1 Window settings: for Uplink DR5, Data Rate Offset 0, SF7 BW500k
- this matches the settings in the Device Profile

The unsuccessful packet is sent with incorrect RX1 Window settings: SF12
- this matches the setting which was in the chirpstack-network-server.toml configuration file.

It seems that even though the device shoudl be following the settings in the Device Profile, these are overrided by the settings from the config file

After changing the config file to match the Device Profile and restarting the process, the downlink is sent with the correct settings.

Although this fix works, it is not a solution: it means that we can never use a different devcie profile.

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | 3.16.0  |
| Network Server      |   3.14.0      |
| Gateway Bridge      |  3.12.0       |
| Chirpstack API      |   3.16.0      |
| Geolocation         |         |
| Concentratord       |         |

mmrein, 2021-12-21:
> It seems that even though the device shoudl be following the settings in the Device Profile, these are overrided by the settings from the config file

Doesn't work that way - device should follow network settings. You can look at device profile as an initial settings so it can start communicating with network, but then the server should send actual settings so the device can adjust.

brocaar, 2022-01-05:
@mmrein  is correct.

The main reason for this is that a LoRaWAN network operator is responsible for the quality of the network and making sure that it is compliant with the regional restrictions and laws.

closed 2022-01-05 by brocaar