brocaar/chirpstack-network-server #572: ACK sent for previous FCnt instead of current one

jorgeig-space, 2022-01-26:
- [X] The issue is present in the latest release.
- [X] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?

Received an UnconfirmedDataDownlink with an ACK for the previous FCnt, instead of the current one

## What did you expect?

Receive an ACK for the current FCnt

## Steps to reproduce this issue

Steps:

1. Join the network
2. Start transmitting confirmed packets, payload is [0,0,0,0,<auto_increment_counter>,0]
3. After a few packets, the latest ACK is duplicated. All previous are correctly acknowledged, although the acknowledgement of the packet previous to the error is not shown in the GUI nor on redis.

I am attaching the app screenshot, end-device application log extract, redis extract (from the gw key) and the network-server log. I am not certain that the problem is in chirpstack because this seems like a very common use case, but looking at redis it looks like a chirpstack issue. Any pointers to where to look next?

[image]

These are log extracts from the end-device application, where we can see that the ACKs are correctly received up to FCnt 6, where it is the same as FCnt 5.
```
// This is the JoinAccept
7.994411 DEBUG parse_with_factory: bytes: [32, 121, 130, 16, 220, 185, 38, 234, 13, 119, 32, 67, 42, 33, 255, 225, 67]
// For each Uplink with FCnt X
30.455295 TRACE lorawan_response: Uplink with FCnt 0
// We receive an UnconfirmedDataDownlink with an ACK
33.970936 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 0, 0, 176, 219, 212, 191]
60.761542 TRACE lorawan_response: Uplink with FCnt 1
64.276677 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 1, 0, 187, 231, 90, 239]
91.076293 TRACE lorawan_response: Uplink with FCnt 2
94.592112 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 2, 0, 44, 128, 182, 181]
121.395793 TRACE lorawan_response: Uplink with FCnt 3
124.911873 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 3, 0, 33, 107, 63, 180]
151.719372 TRACE lorawan_response: Uplink with FCnt 4
155.236551 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 4, 0, 117, 80, 97, 82]
// This is where the issue manifests
182.041880 TRACE lorawan_response: Uplink with FCnt 5
185.558987 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 5, 0, 56, 146, 35, 134] <--------------
212.366891 TRACE lorawan_response: Uplink with FCnt 6
215.883770 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 5, 0, 56, 146, 35, 134] <--------------
```

I am not sure whether the issue is in ChirpStack or the RAK2245 forwarder, but I went into the redis stream and found that the ACK is generated at the wrong time, but strangely enough also generated at the right time although it's not in the redis stream:

Note: the duplicated key "2)" is just my translation of the ASCII bytestring into an array of uint_8.

```
// FCnt 3
 4) 1) "1643198014609-0"
    2) 1) "up"
       2) "\n\x13\x80\xec\xd5\xf4\x01\x00\x03\x00\n\t\"\xc2a\x06u!mv\x9d\x12\x11\b\xd8\xf3\xc6\xce\x01\x1a\t\b}\x10\x0c\x1a\x034/5\x1a>\n\b\xb8'\xeb\xff\xfe\xd7\xe6\xe4(\xfc\xff\xff\xff\xff\xff\xff\xff\xff\x011ffffff,@@\x01Z\x00z\x04\x05\x9b\xa7t\x82\x01\x10\xe2\x0c\xe6d\xe4DE\xbb\xa8\xbeF\x1d}V.\xf8\x88\x01\x02 \x04*\x04\x01\xf4\xd5\xec:\x0c\b\xbe\xec\xc4\x8f\x06\x10\xfb\xf0\xa1\xa0\x02"
       2) [10, 19, 128, 236, 213, 244, 1, 0, 3, 0, 10, 9, 34, 194, 97, 6, 117, 33, 109, 118, 157, 18, 17, 8, 216, 243, 198, 206, 1, 26, 9, 8, 125, 16, 12, 26, 3, 52, 47, 53, 26, 62, 10, 8, 184, 39, 235, 255, 254, 215, 230, 228, 40, 252, 255, 255, 255, 255, 255, 255, 255, 255, 1, 49, 102, 102, 102, 102, 102, 102, 44, 64, 64, 1, 90, 0, 122, 4, 5, 155, 167, 116, 130, 1, 16, 226, 12, 230, 100, 228, 68, 69, 187, 168, 190, 70, 29, 125, 86, 46, 248, 136, 1, 2, 32, 4, 42, 4, 1, 244, 213, 236, 58, 12, 8, 190, 236, 196, 143, 6, 16, 251, 240, 161, 160, 2]
 5) 1) "1643198014931-0"
    2) 1) "down"
       2) "\n\x0c`\xec\xd5\xf4\x01 \x03\x00!k?\xb4\x12$(\xd8\xf3\xc6\xce\x010\n`\x01\x82\x01\x04\x05\x9b\xa7tB\x0b\b}\x10\x0c\x1a\x034/5 \x01r\x04\n\x02\b\x01\x18\xf2\xbb\x02\"\x10\x9d\xf2\xab9\xee6B[\xb8\xc14\x99\x9d0f\b*\b\xb8'\xeb\xff\xfe\xd7\xe6\xe40\x03:\x04\x01\xf4\xd5\xecB\b\xab\xab\xab\xab\xab\xab\xab\xabJ\x0c\b\xbe\xec\xc4\x8f\x06\x10\xef\xb1\xdb\xba\x03"
       2) [10, 12, 96, 236, 213, 244, 1, 32, 3, 0, 33, 107, 63, 180, 18, 36, 40, 216, 243, 198, 206, 1, 48, 10, 96, 1, 130, 1, 4, 5, 155, 167, 116, 66, 11, 8, 125, 16, 12, 26, 3, 52, 47, 53, 32, 1, 114, 4, 10, 2, 8, 1, 24, 242, 187, 2, 34, 16, 157, 242, 171, 57, 238, 54, 66, 91, 184, 193, 52, 153, 157, 48, 102, 8, 42, 8, 184, 39, 235, 255, 254, 215, 230, 228, 48, 3, 58, 4, 1, 244, 213, 236, 66, 8, 171, 171, 171, 171, 171, 171, 171, 171, 74, 12, 8, 190, 236, 196, 143, 6, 16, 239, 177, 219, 186, 3]
// FCnt 4
 6) 1) "1643198044878-0"
    2) 1) "up"
       2) "\n\x13\x80\xec\xd5\xf4\x01\x00\x04\x00\noDO\xa6\x1a;\xda\x84\xeb\xad\x12\x11\b\xd8\xf3\xc6\xce\x01\x1a\t\b}\x10\x0c\x1a\x034/5\x1a>\n\b\xb8'\xeb\xff\xfe\xd7\xe6\xe4(\xfc\xff\xff\xff\xff\xff\xff\xff\xff\x011\x00\x00\x00\x00\x00\x00'@@\x01Z\x00z\x04\ai\xbb\x84\x82\x01\x10-\xaf\xda\xae\x82\rD\xfb\xb7h\xa8\")\x1d\xefO\x88\x01\x02 \x04*\x04\x01\xf4\xd5\xec:\x0c\b\xdc\xec\xc4\x8f\x06\x10\xb0\xff\x85\xa2\x03"
       2) [10, 19, 128, 236, 213, 244, 1, 0, 4, 0, 10, 111, 68, 79, 166, 26, 59, 218, 132, 235, 173, 18, 17, 8, 216, 243, 198, 206, 1, 26, 9, 8, 125, 16, 12, 26, 3, 52, 47, 53, 26, 62, 10, 8, 184, 39, 235, 255, 254, 215, 230, 228, 40, 252, 255, 255, 255, 255, 255, 255, 255, 255, 1, 49, 0, 0, 0, 0, 0, 0, 39, 64, 64, 1, 90, 0, 122, 4, 7, 105, 187, 132, 130, 1, 16, 45, 175, 218, 174, 130, 13, 68, 251, 183, 104, 168, 34, 41, 29, 239, 79, 136, 1, 2, 32, 4, 42, 4, 1, 244, 213, 236, 58, 12, 8, 220, 236, 196, 143, 6, 16, 176, 255, 133, 162, 3]
 7) 1) "1643198045150-0"
    2) 1) "down"
       2) "\n\x0c`\xec\xd5\xf4\x01 \x04\x00uPaR\x12$(\xd8\xf3\xc6\xce\x010\n`\x01\x82\x01\x04\ai\xbb\x84B\x0b\b}\x10\x0c\x1a\x034/5 \x01r\x04\n\x02\b\x01\x18\x9c\xaf\x02\"\x10\x97\x9c\x1f\xa0_\xa0K3\xab\x1bI\x8dP\xd7\x15\xec*\b\xb8'\xeb\xff\xfe\xd7\xe6\xe40\x03:\x04\x01\xf4\xd5\xecB\b\xab\xab\xab\xab\xab\xab\xab\xabJ\x0b\b\xdd\xec\xc4\x8f\x06\x10\xf0\xda\xc8F"
       2) [10, 12, 96, 236, 213, 244, 1, 32, 4, 0, 117, 80, 97, 82, 18, 36, 40, 216, 243, 198, 206, 1, 48, 10, 96, 1, 130, 1, 4, 7, 105, 187, 132, 66, 11, 8, 125, 16, 12, 26, 3, 52, 47, 53, 32, 1, 114, 4, 10, 2, 8, 1, 24, 156, 175, 2, 34, 16, 151, 156, 31, 160, 95, 160, 75, 51, 171, 27, 73, 141, 80, 215, 21, 236, 42, 8, 184, 39, 235, 255, 254, 215, 230, 228, 48, 3, 58, 4, 1, 244, 213, 236, 66, 8, 171, 171, 171, 171, 171, 171, 171, 171, 74, 11, 8, 221, 236, 196, 143, 6, 16, 240, 218, 200, 70]
// FCnt 5 - No ACK seen here but seen on the device!!
[... 158 lines cut ...]
INFO[0270] gateway/mqtt: closing backend                
INFO[0270] gateway/mqtt: unsubscribing from event topic  topic=gateway/+/event/+
INFO[0270] backend/gateway: handling last messages      
INFO[0270] uplink: waiting for pending actions to complete 
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.17.4  |
| Network Server      | v3.15.5        |
| Gateway Bridge      | v3.13.2        |
| Chirpstack API      |   N/A   |
| Geolocation         |  N/A  |
| Concentratord       |  RAK  |

RAK lora_pkt_fwd version:
```
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
```

jorgeig-space, 2022-01-26:
This might also be the cause of this issue: [https://github.com/brocaar/chirpstack-application-server/issues/522](https://github.com/brocaar/chirpstack-application-server/issues/522)

Most stacks will fail silently if they received an invalid frame. In this case the frame is invalid because the MIC will be wrong (as it is for the previous FCnt). The only reason I have the end-node logs above is because I'm also working on the Rust Lorawan stack so I inserted some debug statements.

To clarify, there are no other devices on the network, everything is running on the same host, and the end-device and gateway are in the same room.

brocaar, 2022-01-28:
Do you mean that for two confirmed uplinks (FCnt 5 and FCnt 6), the NS responded with exactly the same downlink payload (containing the ACK)?

```
// This is where the issue manifests
182.041880 TRACE lorawan_response: Uplink with FCnt 5
185.558987 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 5, 0, 56, 146, 35, 134] <--------------
212.366891 TRACE lorawan_response: Uplink with FCnt 6
215.883770 DEBUG parse_with_factory: bytes: [96, 236, 213, 244, 1, 32, 5, 0, 56, 146, 35, 134] <--------------
```

Please note that (for LoRaWAN 1.0), the ACK does not contain the frame-counter that was acknowledged. For LoRaWAN 1.1 the ACK uses the uplink frame-counter in the MIC calculation.

Note that the NS increments the downlink frame-counter after receiving a TX ACK from the gateway. In case the gateway does transmit the downlink, but does not send a TXACK (or it did send but it did not reach the NS), the NS will consider the downlink as not transmitted and the downlink frame-counter will not increment and thus will be used again for the next downlink.

This could have happened:

1. Device sends confirmed up with fcnt up: 5
2. NS responds with ACK using fcnt down: 5 (no TX ACK from gateway)
3. Device sends confirmed up with fcnt up: 6
4. NS responds with ACK using fcnt down: 5 (TX ACK from gateway)

In this case the downlink from 2) will not appear in the Redis Stream as it is considered as never been sent by the gateway.

Please note, it is perfectly fine that the confirmed uplink has fcnt 6, and the downlink acknowledgement has fcnt 5. The device receives an ACK, but it does not contain the uplink fcnt to which the ACK relates. That is why in LoRaWAN 1.1 this was included in the MIC validation.

jorgeig-space, 2022-01-29:
Many thanks for taking the time to look into this and pointing out the ACK mechanism. That makes things much clearer. Clearly this is not an NS issue.

It appears there is some issue in RAK's `lora_pkt_fwd`, since the ACK is transmitted (seen by the end-device), but no txack is given back to the NS... I'll close the issue here and post a solution when I find it. If you have any ideas on how to diagnose that, they'll be much appreciated! Thanks.

jorgeig-space, 2022-03-02:
Hi, sorry to beat the dead horse with this but I've been taking a closer look at logs and it appears the NS is indeed receiving TX ACK from the gateway. Redis still doesn't show any downlink message right before the problem, and the device is still seeing the "duplicated" ACK.

In this latest test, FCnt 0 is ACK'd properly, FCnt 1 is ACK'd as seen by the device and the NS shows the downlink is ACK'd by the gateway, but the NS seems to ignore that, as it sends an ACK with FCnt 1 in the next uplink. The big issue is that the ACK is always correctly transmitted, it's just that the NS doesn't seem to know about it (but the log says otherwise).

Any pointer on where to look? I can only see one difference between the good ACKs and the one that appears to be ignored: at the end of each FCnt block, there is a message in the NS log `msg="sent downlink meta-data to network-controller"` which is not present for FCnt 1.

The messages `storage: downlink-frame saved` seem to indicate that we are on the `OnError` branch of [handleDownlinkTXAckTasks](https://github.com/brocaar/chirpstack-network-server/blob/2aea7096f5036030823517fcce5a0c97021becc5/internal/downlink/ack/ack.go#L51), but I see no errors on the AS or anywhere else.

=========== Logs

View on application:
[image]

NS Log:
```
JOIN
ns: time="2022-03-02T03:22:41.287768752Z" level=info msg="gateway/mqtt: uplink frame received" gateway_id=b827ebfffed7e6e4 uplink_id=34d5f339-a1c4-4374-8fb1-1673b2c234d6
ns: time="2022-03-02T03:22:41.506105913Z" level=info msg="uplink: frame(s) collected" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 mtype=JoinRequest uplink_ids="[34d5f339-a1c4-4374-8fb1-1673b2c234d6]"
ns: time="2022-03-02T03:22:41.661447048Z" level=info msg="lorawan/backend: finished backend api call" message_type=JoinReq protocol_version=1.0 receiver_id=0000000000000000 result_code=Success sender_id=000000 transaction_id=187779681
ns: time="2022-03-02T03:22:41.669978781Z" level=info msg="sent uplink meta-data to network-controller" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 dev_eui=3236500700160026
ns: time="2022-03-02T03:22:41.677317551Z" level=info msg="device-queue flushed" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 dev_eui=3236500700160026
ns: time="2022-03-02T03:22:41.687746224Z" level=info msg="device-session saved" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 dev_addr=01f8eea2 dev_eui=3236500700160026
ns: time="2022-03-02T03:22:41.705809659Z" level=info msg="device-activation created" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 dev_eui=3236500700160026 id=1063
ns: time="2022-03-02T03:22:41.716904311Z" level=info msg="device updated" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 dev_eui=3236500700160026
ns: time="2022-03-02T03:22:41.726039025Z" level=info msg="gateway/mqtt: publishing gateway command" command=down downlink_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 gateway_id=b827ebfffed7e6e4 qos=0 topic=gateway/b827ebfffed7e6e4/command/down
ns: time="2022-03-02T03:22:41.745245424Z" level=info msg="storage: downlink-frame saved" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 token=50642
------> ns: time="2022-03-02T03:22:41.761914901Z" level=info msg="backend/gateway: downlink tx acknowledgement received" downlink_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535 gateway_id=b827ebfffed7e6e4
ns: time="2022-03-02T03:22:41.781753280Z" level=info msg="sent downlink meta-data to network-controller" ctx_id=795fcae2-a24a-4f49-b8d3-60e0ca3f0535
FCnt 0
ns: time="2022-03-02T03:22:49.236545186Z" level=info msg="gateway/mqtt: uplink frame received" gateway_id=b827ebfffed7e6e4 uplink_id=3d626ee7-52c4-4e21-95ed-442f665a9918
ns: time="2022-03-02T03:22:49.465267739Z" level=info msg="uplink: frame(s) collected" ctx_id=79cb52e5-294d-4041-bc09-855a5adf21dd mtype=ConfirmedDataUp uplink_ids="[3d626ee7-52c4-4e21-95ed-442f665a9918]"
ns: time="2022-03-02T03:22:49.569812964Z" level=info msg="sent uplink meta-data to network-controller" ctx_id=79cb52e5-294d-4041-bc09-855a5adf21dd dev_eui=3236500700160026
ns: time="2022-03-02T03:22:49.581033451Z" level=info msg="device gateway rx-info meta-data saved" ctx_id=79cb52e5-294d-4041-bc09-855a5adf21dd dev_eui=3236500700160026
ns: time="2022-03-02T03:22:49.602659464Z" level=info msg="device-session saved" ctx_id=79cb52e5-294d-4041-bc09-855a5adf21dd dev_addr=01f8eea2 dev_eui=3236500700160026
ns: time="2022-03-02T03:22:49.750499711Z" level=info msg="finished client unary call" ctx_id=79cb52e5-294d-4041-bc09-855a5adf21dd grpc.code=OK grpc.ctx_id=8aae7825-2f8c-42ea-8c79-ff4a52488b6f grpc.duration=163.189546ms grpc.method=HandleUplinkData grpc.service=as.ApplicationServerService span.kind=client system=grpc
ns: time="2022-03-02T03:22:49.776026545Z" level=info msg="gateway/mqtt: publishing gateway command" command=down downlink_id=79cb52e5-294d-4041-bc09-855a5adf21dd gateway_id=b827ebfffed7e6e4 qos=0 topic=gateway/b827ebfffed7e6e4/command/down
ns: time="2022-03-02T03:22:49.813452835Z" level=info msg="device-session saved" ctx_id=79cb52e5-294d-4041-bc09-855a5adf21dd dev_addr=01f8eea2 dev_eui=3236500700160026
-----> ns: time="2022-03-02T03:22:49.822347429Z" level=info msg="backend/gateway: downlink tx acknowledgement received" downlink_id=79cb52e5-294d-4041-bc09-855a5adf21dd gateway_id=b827ebfffed7e6e4
ns: time="2022-03-02T03:22:49.847309289Z" level=info msg="storage: downlink-frame saved" ctx_id=79cb52e5-294d-4041-bc09-855a5adf21dd token=31179
[... cut ...] 2,"fPort":10,"data":"1uIeYgMA","objectJSON":"{\"fPort\":10,\"frame_number\":3,\"size\":0,\"timestamp\":0}","tags":{},"confirmedUplink":true,"devAddr":"Afjuog==","publishedAt":"2022-03-02T03:22:57.349608035Z","deviceProfileID":"8917dda4-c938-4449-a5d1-45e5bcfda112","deviceProfileName":"Munin Sensor"}

FCnt 2 ACK to NS (although the Down command now contains FCnt_Down 1, as if the previous one wasn't ACK'd)
2022-03-02T03:22:57+0000 gateway/b827ebfffed7e6e4/event/ack 
?'????????"?X?)?mG??´?R??*
```

jorgeig-space, 2022-03-05:
Updated title and description to match the problem more closely.

jorgeig-space, 2022-03-05:
I've also just tried using the Chirpstack OS image (full) for Raspberry Pi Zero W. The problem still persists using the Chirpstack OS image, configured for EU433.

jorgeig-space, 2022-03-11:
It turns out this only happens when running the full chirpstack stack in a Pi Zero W. When the same stack runs on a separate computer, with the RPi running the packet forwarder, everything works normally. Also tried with TTN and everything worked.

I am closing the issue, as it's not a bug per se, but if you have some free time I'll be happy to dig with you as to why this happens without any error messages. If you need a RPi Zero let me know :)

closed 2022-03-11 by jorgeig-space