brocaar/chirpstack-network-server #576: Device Logs showing error message.

mudassirahmed395, 2022-02-21:
- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
When I enter the device logs in chirpStack the device shows log errors.  Attached below is the picture of it.
[image]

## What did you expect?
I expect to have logs without error message as some of the log messages does not show any error message attached below.
## Steps to reproduce this issue

Steps:

1. This issues gets produce when I open the "Device Data". Sometime it produces error sometime logs are working fine so it is hard to tell what exactly is the procedure to reproduce this error.
[image]

## Could you share your log output?
[
    {
        "type": "error",
        "payload": {
            "applicationID": "232",
            "applicationName": "brighter-bins-test",
            "deviceName": "production-laser-0080e11500073d52",
            "devEUI": "AIDhFQAHPVI=",
            "type": "UPLINK_FCNT_RETRANSMISSION",
            "error": "frame-counter did not increment",
            "fCnt": 28,
            "tags": {},
            "publishedAt": "2022-02-21T10:38:04.688143501Z"
        }
    },
    {
        "type": "up",
        "payload": {
            "applicationID": "232",
            "applicationName": "brighter-bins-test",
            "deviceName": "production-laser-0080e11500073d52",
            "devEUI": "AIDhFQAHPVI=",
            "rxInfo": [
                {
                    "gatewayID": "/MI9//4LbxM=",
                    "time": null,
                    "timeSinceGPSEpoch": null,
                    "rssi": -88,
                    "loRaSNR": 8.5,
                    "channel": 1,
                    "rfChain": 1,
                    "board": 0,
                    "antenna": 0,
                    "location": {
                        "latitude": 53.74952285419751,
                        "longitude": -1.4503240585327148,
                        "altitude": 0,
                        "source": "UNKNOWN",
                        "accuracy": 0
                    },
                    "fineTimestampType": "NONE",
                    "context": "7k1gow==",
                    "uplinkID": "N/JtCSpySgGhQlVODcR77g==",
                    "crcStatus": "CRC_OK"
                }
            ],
            "txInfo": {
                "frequency": 868300000,
                "modulation": "LORA",
                "loRaModulationInfo": {
                    "bandwidth": 125,
                    "spreadingFactor": 7,
                    "codeRate": "4/5",
[... 202 lines cut ...]
            "deviceName": "production-laser-0080e11500073d52",
            "devEUI": "AIDhFQAHPVI=",
            "type": "UPLINK_FCNT_RETRANSMISSION",
            "error": "frame-counter did not increment",
            "fCnt": 25,
            "tags": {},
            "publishedAt": "2022-02-21T09:08:04.564398874Z"
        }
    }
]

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

liangyuanpeng, 2022-03-10:
Seems like your sensor have not correct increment the `fcnt`.

mudassirahmed395, 2022-04-01:
But device work fine in TTN Actiltiy Proimous.

brocaar, 2022-04-28:
This is because of https://github.com/brocaar/chirpstack-network-server/issues/560.

closed 2022-04-28 by brocaar