brocaar/chirpstack-network-server #580: Tests stuck

sergorl, 2022-04-18:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
Tests stuck.

## What did you expect?
Test should be passed or not.

## Steps to reproduce this issue

Steps:

1. `make test`

## Could you share your log output?

```shell
Running tests
adr/structs.go:97:1: exported method HandlerRPCServer.ID should have comment or be unexported
adr/structs.go:103:1: exported method HandlerRPCServer.Name should have comment or be unexported
adr/structs.go:109:1: exported method HandlerRPCServer.Handle should have comment or be unexported
adr/structs.go:120:1: exported method HandlerRPC.ID should have comment or be unexported
adr/structs.go:126:1: exported method HandlerRPC.Name should have comment or be unexported
adr/structs.go:132:1: exported method HandlerRPC.Handle should have comment or be unexported
adr/structs.go:144:1: exported method HandlerPlugin.Server should have comment or be unexported
adr/structs.go:148:1: exported method HandlerPlugin.Client should have comment or be unexported
examples/adr-plugin/main.go:10:1: comment on exported type Handler should be of the form "Handler ..." (with optional leading article)
internal/api/ns/api.go:20:1: exported function Setup should have comment or be unexported
internal/api/ns/network_server.go:1029:1: receiver name c should be consistent with previous receiver name n for NetworkServerAPI
internal/api/roaming/roaming.go:366:9: if block ends with a return statement, so drop this else and outdent its block
internal/backend/controller/nop_client.go:20:1: comment on exported method NopNetworkControllerClient.HandleDownlinkMetaData should be of the form "HandleDownlinkMetaData ..."
internal/backend/gateway/amqp/amqp.go:77:1: exported method Backend.SendTXPacket should have comment or be unexported
internal/backend/gateway/amqp/amqp.go:96:1: exported method Backend.SendGatewayConfigPacket should have comment or be unexported
internal/backend/gateway/amqp/amqp.go:108:1: exported method Backend.RXPacketChan should have comment or be unexported
internal/backend/gateway/amqp/amqp.go:112:1: exported method Backend.StatsPacketChan should have comment or be unexported
internal/backend/gateway/amqp/amqp.go:116:1: exported method Backend.DownlinkTXAckChan should have comment or be unexported
internal/backend/gateway/amqp/amqp.go:120:1: exported method Backend.Close should have comment or be unexported
internal/backend/gateway/amqp/amqp.go:397:21: error strings should not be capitalized or end with punctuation or a newline
internal/backend/gateway/amqp/channel_pool.go:67:8: should omit values from range; this loop is equivalent to `for range ...`
internal/backend/gateway/azureiothub/azureiothub.go:154:1: exported method Backend.SendTXPacket should have comment or be unexported
internal/backend/gateway/azureiothub/azureiothub.go:173:1: exported method Backend.SendGatewayConfigPacket should have comment or be unexported
internal/backend/gateway/azureiothub/azureiothub.go:185:1: exported method Backend.RXPacketChan should have comment or be unexported
internal/backend/gateway/azureiothub/azureiothub.go:189:1: exported method Backend.StatsPacketChan should have comment or be unexported
internal/backend/gateway/azureiothub/azureiothub.go:193:1: exported method Backend.DownlinkTXAckChan should have comment or be unexported
internal/backend/gateway/azureiothub/azureiothub.go:197:1: exported method Backend.Close should have comment or be unexported
internal/config/config.go:241:6: exported type RoamingServer should have comment or be unexported
internal/config/config.go:256:6: exported type DefaultRoamingServer should have comment or be unexported
internal/config/config.go:270:6: exported type KEK should have comment or be unexported
internal/downlink/ack/ack_roaming_hns.go:10:1: comment on exported function HandleRoamingTxAck should be of the form "HandleRoamingTxAck ..."
internal/downlink/data/data.go:791:11: if block ends with a return statement, so drop this else and outdent its block
internal/downlink/join/join.go:94:21: error strings should not be capitalized or end with punctuation or a newline
internal/downlink/roaming/roaming.go:50:1: exported function EmitPRDownlink should have comment or be unexported
internal/gateway/stats/stats.go:22:5: exported var ErrAbort should have comment or be unexported
internal/logging/logging.go:53:1: exported function UnaryClientCtxIDInterceptor should have comment or be unexported
internal/migrations/code/test/code_test.go:30:1: receiver name ts should be consistent with previous receiver name b for MigrateTestSuite
internal/roaming/dl_metadata.go:11:1: comment on exported function DLMetaDataToUplinkRXInfoSet should be of the form "DLMetaDataToUplinkRXInfoSet ..."
internal/roaming/ul_metadata.go:17:1: exported function ULMetaDataToTXInfo should have comment or be unexported
internal/roaming/ul_metadata.go:35:1: exported function ULMetaDataToRXInfo should have comment or be unexported
internal/roaming/ul_metadata.go:86:1: exported function RecvTimeFromRXInfo should have comment or be unexported
internal/roaming/ul_metadata.go:101:1: exported function RXInfoToGWInfo should have comment or be unexported
internal/uplink/join/join_roaming_sns.go:12:1: comment on exported function HandleStartPRHNS should be of the form "HandleStartPRHNS ..."
?       github.com/brocaar/chirpstack-network-server/v3/adr     [no test files]
?       github.com/brocaar/chirpstack-network-server/v3/cmd/chirpstack-network-server   [no test files]
?       github.com/brocaar/chirpstack-network-server/v3/cmd/chirpstack-network-server/cmd       [no test files]
?       github.com/brocaar/chirpstack-network-server/v3/examples/adr-plugin     [no test files]
=== RUN   TestADR
=== RUN   TestADR/GetHandler
--- PASS: TestADR (0.00s)
    --- PASS: TestADR/GetHandler (0.00s)
[... 69 lines cut ...]
    --- PASS: TestLRFHSSHandler/ID (0.00s)
    --- PASS: TestLRFHSSHandler/Handle_without_LR-FHSS_enabled (0.00s)
    --- PASS: TestLRFHSSHandler/Handle (0.00s)
        --- PASS: TestLRFHSSHandler/Handle/adr_disabled (0.00s)
        --- PASS: TestLRFHSSHandler/Handle/max_dr._prevents_lr-fhss (0.00s)
        --- PASS: TestLRFHSSHandler/Handle/switch_to_dr_10 (0.00s)
        --- PASS: TestLRFHSSHandler/Handle/switch_to_dr_11 (0.00s)
PASS
coverage: 78.0% of statements
ok      github.com/brocaar/chirpstack-network-server/v3/internal/adr    0.009s  coverage: 78.0% of statements
?       github.com/brocaar/chirpstack-network-server/v3/internal/api/client/asclient    [no test files]
=== RUN   TestNetworkServerAPINew

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  |3.17.6 |
| Network Server      | 3.16.1 |
| Gateway Bridge      |3.13.2|
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2022-04-18:
Are you using the provided docker-compose environment?

sergorl, 2022-04-18:
@sergorl , no, I use `master` branch from this repository. Then I build it and run tests.

brocaar, 2022-04-18:
Please use the provided docker-compose environment. It comes with all required dependencies already setup for you. It is very likely that the tests are hanging because of failing db connection etc... E.g.

```
# Enter development shell
docker-compose run --rm chirpstack-network-server bash

# Within development shell
make test
```

closed 2022-04-18 by brocaar