brocaar/chirpstack-network-server #586: No LORAWAN Frames: level=error msg="gateway: handle gateway stats error" error="update gateway state error: object does not exist"

sergorl, 2022-05-06:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
I use this [gateway](https://docs.rakwireless.com/Product-Categories/WisGate/RAK7248/Datasheet/#overview) without any connected devices.
In application-server `Status seen` is `a few seconds ago`, `Gateway activity` is `None`, `GPS coordinates` is `(0, 0)`, `Live LORAWAN Frames` is `None`.

## What did you expect?
Right status in application-server: correct GPScoordinates and LORAWAN frames.

## Steps to reproduce this issue

Steps:

1. Run [docker-chirpstack](https://github.com/brocaar/chirpstack-docker) by `docker-compose up`
2. Create gateway in application-server.
3.
4.

## Could you share your log output?

```shell
time="2022-05-06T10:57:20.073539229Z" level=info msg="starting ChirpStack Network Server" band=EU868 docs="https://www.chirpstack.io/" net_id=000000 version=
time="2022-05-06T10:57:20.077905827Z" level=info msg="storage: setting up storage module"
time="2022-05-06T10:57:20.077929453Z" level=info msg="storage: setting up Redis client"
time="2022-05-06T10:57:20.07898924Z" level=info msg="storage: connecting to PostgreSQL"
time="2022-05-06T10:57:20.13645922Z" level=info msg="storage: applying PostgreSQL data migrations"
time="2022-05-06T10:57:20.162216879Z" level=info msg="gateway/mqtt: connecting to mqtt broker" server="tcp://mosquitto:1883"
time="2022-05-06T10:57:20.16459562Z" level=info msg="api: starting network-server api server" bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
time="2022-05-06T10:57:20.164829214Z" level=info msg="starting downlink device-queue scheduler"
time="2022-05-06T10:57:20.164846376Z" level=info msg="starting multicast scheduler"
time="2022-05-06T10:57:20.16916317Z" level=info msg="backend/gateway: connected to mqtt server"
time="2022-05-06T10:57:20.171068114Z" level=info msg="gateway/mqtt: subscribing to gateway event topic" qos=0 topic=gateway/+/event/+
time="2022-05-06T10:57:46.910962741Z" level=info msg="finished unary call with code OK" ctx_id=a97c0c3c-eb8c-48e9-830f-189628d0072e grpc.code=OK grpc.method=GetGatewayProfile grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:57:46Z" grpc.time_ms=5.032 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:57:55.634779455Z" level=info msg="finished unary call with code OK" ctx_id=c4392ad7-d18a-463b-b950-1b1f122b617e grpc.code=OK grpc.method=GetGateway grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:57:55Z" grpc.time_ms=6.272 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:57:58.452940993Z" level=info msg="finished unary call with code OK" ctx_id=34e4bffb-1629-49cd-8b27-2a4780902134 grpc.code=OK grpc.method=GetGatewayProfile grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:57:58Z" grpc.time_ms=1.383 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:57:58.472481087Z" level=info msg="finished unary call with code OK" ctx_id=8ab8846b-28c1-4e54-a0f9-4c6a0b6694fe grpc.code=OK grpc.method=GetServiceProfile grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:57:58Z" grpc.time_ms=2.663 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:58:03.770056614Z" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=e45f01fffe14ccbf stats_id=8a5d3a35-d7ce-4fd7-8daa-c3cdbdf8d527
time="2022-05-06T10:58:03.771701283Z" level=error msg="gateway: handle gateway stats error" ctx_id=8a5d3a35-d7ce-4fd7-8daa-c3cdbdf8d527 error="update gateway state error: object does not exist"
time="2022-05-06T10:58:05.177095977Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamFrameLogsForGateway grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:58:01Z" grpc.time_ms=3465.271 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:58:06.625644469Z" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=e45f01fffe14ccfe stats_id=198dfc0c-5167-4d6f-bd6f-0367c4fa18d3
time="2022-05-06T10:58:06.629585586Z" level=info msg="storage: gateway state updated" ctx_id=198dfc0c-5167-4d6f-bd6f-0367c4fa18d3 gateway_id=e45f01fffe14ccfe
time="2022-05-06T10:58:06.632363666Z" level=warning msg="creating insecure application-server client" server="chirpstack-application-server:8001"
time="2022-05-06T10:58:06.642720914Z" level=info msg="finished client unary call" ctx_id=198dfc0c-5167-4d6f-bd6f-0367c4fa18d3 grpc.code=OK grpc.ctx_id=599ed36b-a238-4f68-8434-890ac75140c9 grpc.duration=8.401583ms grpc.method=HandleGatewayStats grpc.service=as.ApplicationServerService span.kind=client system=grpc
time="2022-05-06T10:58:33.77209287Z" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=e45f01fffe14ccbf stats_id=48cb2767-2b3c-4131-a32f-4fe47010a472
time="2022-05-06T10:58:33.773250731Z" level=error msg="gateway: handle gateway stats error" ctx_id=48cb2767-2b3c-4131-a32f-4fe47010a472 error="update gateway state error: object does not exist"
time="2022-05-06T10:58:36.632609925Z" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=e45f01fffe14ccfe stats_id=f01a777e-75f7-4810-97bc-6fb7139a6d68
time="2022-05-06T10:58:36.634864186Z" level=info msg="storage: gateway state updated" ctx_id=f01a777e-75f7-4810-97bc-6fb7139a6d68 gateway_id=e45f01fffe14ccfe
time="2022-05-06T10:58:36.64197737Z" level=info msg="finished client unary call" ctx_id=f01a777e-75f7-4810-97bc-6fb7139a6d68 grpc.code=OK grpc.ctx_id=ad91f17d-4f80-43c6-9cf7-d806fcbed532 grpc.duration=5.627705ms grpc.method=HandleGatewayStats grpc.service=as.ApplicationServerService span.kind=client system=grpc
time="2022-05-06T10:59:00.520923253Z" level=info msg="finished unary call with code OK" ctx_id=770ffec8-9e24-4846-abc9-dbafdbf997fb grpc.code=OK grpc.method=GetGateway grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:59:00Z" grpc.time_ms=1.807 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:59:03.529176286Z" level=info msg="finished unary call with code OK" ctx_id=124a6084-ec42-43e2-ae33-37a7f084c801 grpc.code=OK grpc.method=GetServiceProfile grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:59:03Z" grpc.time_ms=1.248 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:59:03.667266252Z" level=info msg="finished unary call with code OK" ctx_id=e93e1bd1-c34a-4e13-8944-de171d3b4935 grpc.code=OK grpc.method=GetGatewayProfile grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:59:03Z" grpc.time_ms=1.426 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:59:03.773893891Z" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=e45f01fffe14ccbf stats_id=bde69211-f028-4417-88c9-fdde210578fa
time="2022-05-06T10:59:03.775056162Z" level=error msg="gateway: handle gateway stats error" ctx_id=bde69211-f028-4417-88c9-fdde210578fa error="update gateway state error: object does not exist"
time="2022-05-06T10:59:06.631545456Z" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=e45f01fffe14ccfe stats_id=9f2fc61b-5184-4957-b6e6-cf49bc9a74ea
time="2022-05-06T10:59:06.633671238Z" level=info msg="storage: gateway state updated" ctx_id=9f2fc61b-5184-4957-b6e6-cf49bc9a74ea gateway_id=e45f01fffe14ccfe
time="2022-05-06T10:59:06.640293579Z" level=info msg="finished client unary call" ctx_id=9f2fc61b-5184-4957-b6e6-cf49bc9a74ea grpc.code=OK grpc.ctx_id=cd86c7bc-7901-4e17-befa-490234b73699 grpc.duration=5.341726ms grpc.method=HandleGatewayStats grpc.service=as.ApplicationServerService span.kind=client system=grpc
time="2022-05-06T10:59:08.853991855Z" level=info msg="finished streaming call with code OK" grpc.code=OK grpc.method=StreamFrameLogsForGateway grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:59:07Z" grpc.time_ms=1265.902 peer.address="172.19.0.7:54974" span.kind=server system=grpc
time="2022-05-06T10:59:31.032261223Z" level=info msg="finished unary call with code OK" ctx_id=01c44d7c-9ca8-4808-b71e-6edf940299b7 grpc.code=OK grpc.method=GetVersion grpc.service=ns.NetworkServerService grpc.start_time="2022-05-06T10:59:31Z" grpc.time_ms=0.06 peer.address="172.19.0.7:54974" span.kind=server system=grpc
[... cut ...]
time="2022-05-06T11:02:38.098591235Z" level=debug msg="sql query executed" args="[100 371075h2m58.096787259s 2022-05-06 11:02:38.096790311 +0000 UTC m=+4.481217323]" duration=1.790273ms query="\n        select\n            d.*\n        from\n            device d\n        where\n\t\t\td.mode in ('B', 'C')\n            -- we want devices with queue items\n   
```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

winnieXY, 2022-06-21:
I upgraded my gatway from debian buster to debian bullseye yesterday and after that the gateway/network server stopped working. So far I'm not able to find the issue - but i get a very similiar error message:

```Jun 21 08:54:32 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T08:54:32.369130171+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=a5ddae55-7545-4409-ba29-84ed70627b1c error="update gateway state error: object does not exist"

Jun 21 08:55:02 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T08:55:02.37173928+01:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=b827ebfffeaXXXX stats_id=95ba73ce-1cd6-499c-8163-c2f03bf0cc24

Jun 21 08:55:02 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T08:55:02.375361748+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=95ba73ce-1cd6-499c-8163-c2f03bf0cc24 error="update gateway state error: object does not exist"

Jun 21 08:55:32 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T08:55:32.364950213+01:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=b827ebfffeaXXXX stats_id=a848aa4a-d42f-40f9-bb08-5613e9d81118

Jun 21 08:55:32 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T08:55:32.369636422+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=a848aa4a-d42f-40f9-bb08-5613e9d81118 error="update gateway state error: object does not exist"

Jun 21 08:56:02 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T08:56:02.365769431+01:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=b827ebfffeaXXXX stats_id=9b9dd315-b759-45e1-9751-41c1d7ba20e4

Jun 21 08:56:02 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T08:56:02.370086998+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=9b9dd315-b759-45e1-9751-41c1d7ba20e4 error="update gateway state error: object does not exist"
```
My setup looks like this: raspberry pi with rasbian (now bullseye):

root@rak-gateway:/opt/ttn-gateway/packet_forwarder/lora_pkt_fwd# dpkg -l | grep chirpstack
ii  chirpstack-application-server        3.17.6                           armhf        ChirpStack Application Server
ii  chirpstack-gateway-bridge            3.14.0                           armhf        ChirpStack Gateway Bridge
ii  chirpstack-network-server            3.16.2                           armhf        ChirpStack Network Server

I've already created new postgresql databases for chirpstack_as and chirpstack_ns (switched to chirpstack_as2 and chirpstack_ns2). I can reproduce this error even with this fresh setup.

Within the applicaton-server frontend the gatway is always listed as "Never" - so the communication is no longer working - although the gateway-bridge and the semtech_udp  concentrator are up and running:

lora_pkt_fwd output:
```
##### 2022-06-21 08:01:27 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 1
# CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 2665592
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 1655798487 sec)
# no valid GPS coordinates available yet
##### END #####
 ```

gateway-bridge output
``` 
Jun 21 09:01:02 rak-gateway chirpstack-gateway-bridge[6788]: time="2022-06-21T09:01:02.37340725+01:00" level=info msg="integration/mqtt: publishing event" event=stats qos=0 stats_id=5c182a54-6fef-431b-84ba-5653a89e70a5 topic=gateway/b827ebfffeaXXXX/event/stats
Jun 21 09:01:27 rak-gateway chirpstack-gateway-bridge[6788]: time="2022-06-21T09:01:27.193790161+01:00" level=info msg="integration/mqtt: publishing event" event=stats qos=0 stats_id=b5801d7f-dab0-42ef-9113-16774c7836c8 topic=gateway/b827ebfffeaXXXX/event/stats
Jun 21 09:01:32 rak-gateway chirpstack-gateway-bridge[6788]: time="2022-06-21T09:01:32.370461048+01:00" level=info msg="integration/mqtt: publishing event" event=stats qos=0 stats_id=7e53b92b-644e-4d64-98ca-b06dcc72d2cd topic=gateway/b827ebfffeaXXXX/event/stats
Jun 21 09:02:02 rak-gateway chirpstack-gateway-bridge[6788]: time="2022-06-21T09:02:02.36933739+01:00" level=info msg="integration/mqtt: publishing event" event=stats qos=0 stats_id=409220ae-5e5e-4910-a041-ce0de90f2add topic=gateway/b827ebfffeaXXXX/event/stats
```

Any help appreciated getting it back running.

winnieXY, 2022-06-21:
FYI I'm going through this debug help now:
https://www.chirpstack.io/project/guides/connect-gateway/

``` 
INFO: [down] PULL_ACK received in 0 ms
##### 2022-06-21 08:07:35 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (113 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 89660477
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# Invalid time reference (age: 1655798855 sec)
# no valid GPS coordinates available yet
##### END #####
JSON up: {"stat":{"time":"2022-06-21 08:07:35 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
INFO: [up] PUSH_ACK received in 1 ms
```
=> Package Forwarder is working

Subscribing to topic for gateway-bridge is working:
``` 
sudo journalctl -f -n 1000 -u chirpstack-gateway-bridge | grep subscribing
Jun 21 08:15:32 rak-gateway chirpstack-gateway-bridge[6788]: time="2022-06-21T08:15:32.363117187+01:00" level=info msg="integration/mqtt: subscribing to topic" qos=0 topic="gateway/b827ebfffea341b8/command/#"
```

Receiving Data on gateway-bridge is working:
```
Jun 21 09:06:08 rak-gateway chirpstack-gateway-bridge[6788]: time="2022-06-21T09:06:08.412908762+01:00" level=info msg="integration/mqtt: publishing event" event=up qos=0 topic=gateway/b827ebfffea341b8/event/up uplink_id=12b670b1-0957-46c6-9805-41911ba75073
Jun 21 09:06:32 rak-gateway chirpstack-gateway-bridge[6788]: time="2022-06-21T09:06:32.372681453+01:00" level=info msg="integration/mqtt: publishing event" event=stats qos=0 stats_id=99476335-fd93-4425-b54e-9e74118a39eb topic=gateway/b827ebfffea341b8/event/stats
```

I get the stats also within mosquitto_sub:
```
root@rak-gateway:/opt/ttn-gateway/packet_forwarder/lora_pkt_fwd# mosquitto_sub -v -t "gateway/#"
gateway/b827ebfffef614ed/state/conn                                                                                                                                                                           ▒'▒▒▒▒▒
gateway/b827ebfffea341b8/state/conn                                                                                                                                                                           ▒'▒▒▒▒A▒
gateway/b827ebfffea341b8/event/stats                                                                                                                                                                          ▒'▒▒▒▒A▒▒ŕJ192.168.68.120Z▒▒φ▒L*▒▒▒fb▒▒
gateway/b827ebfffea341b8/event/stats                                                                                                                                                                          ▒'▒▒▒▒A▒▒ŕJ192.168.68.120Zdg▒▒V▒A▒▒▒6▒▒▒▒
gateway/b827ebfffea341b8/event/stats                                                                                                                                                                          ▒▒4▒▒▒A▒▒ŕJ192.168.68.120Z▒y▒▒▒▒Os▒▒
gateway/b827ebfffea341b8/event/stats                                                                                                                                                                          ▒'▒▒▒▒A▒▒ŕJ192.168.68.120Z▒*ņ▒EL▒▒蹺▒']
```
(Sorry - binary is not shown here somehow). 

The network-server receives the gateway stats packet:
```
Jun 21 09:17:02 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T09:17:02.385186343+01:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=b827ebfffea341b8 stats_id=9c490976-4c61-4476-8581-1084077f7a34
Jun 21 09:17:02 rak-gateway chirpstack-network-server[6803]: time="2022-06-21T09:17:02.38887053+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=9c490976-4c61-4476-8581-1084077f7a34 error="update gateway state error: object does not exist"
```
Same output with debug level increased - last entry within the post - warning long snipplet!

```
Jun 21 09:20:02 rak-gateway chirpstack-network-server[8052]: time="2022-06-21T09:20:02.383846941+01:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=b827ebfffea341b8 stats_id=710f20e1-1a20-419a-b816-254392708ab5
Jun 21 09:20:02 rak-gateway chirpstack-network-server[8052]: time="2022-06-21T09:20:02.389909649+01:00" level=debug msg="sql query executed" args="[b827ebfffea341b8 2022-06-21 09:20:02.384181001 +0100 BST m=+85.815392995 2022-06-21 09:20:02.384181001 +0100 BST m=+85.815392995 <nil> <nil>]" duration=5.653701ms query="\n\t\tupdate gateway set\n\t\t\tfirst_seen_at = coalesce(first_seen_at, $2),\n\t\t\tlast_seen_at = $3,\n\t\t\tlocation = coalesce($4, location),\n\t\t\taltitude = coalesce($5, altitude)\n\t\twhere\n\t\t\tgateway_id = $1"
Jun 21 09:20:02 rak-gateway chirpstack-network-server[8052]: time="2022-06-21T09:20:02.390547092+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=710f20e1-1a20-419a-b816-254392708ab5 error="update gateway state error: object does not exist"
Jun 21 09:20:02 rak-gateway chirpstack-network-server[8052]: time="2022-06-21T09:20:02.720600134+01:00" level=debug msg="running class-b / class-c scheduler batch" ctx_id=3fab9bd5-bc1e-45a3-88df-d41e78b87174
[... cut ...] 47109ms query="\n\t\tupdate gateway set\n\t\t\tfirst_seen_at = coalesce(first_seen_at, $2),\n\t\t\tlast_seen_at = $3,\n\t\t\tlocation = coalesce($4, location),\n\t\t\taltitude = coalesce($5, altitude)\n\t\twhere\n\t\t\tgateway_id = $1"
Jun 21 09:20:32 rak-gateway chirpstack-network-server[8052]: time="2022-06-21T09:20:32.384567776+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=d6c17ffd-bbb0-4b59-861b-05e7663af57b error="update gateway state error: object does not exist"

```

sagar-patel-sls, 2022-06-21:
Hi @winnieXY 

`Jun 21 09:20:32 rak-gateway chirpstack-network-server [8052]: time="2022-06-21T09:20:32.384567776+01:00" level=error msg="gateway: handle gateway stats error" ctx_id=d6c17ffd-bbb0-4b59-861b-05e7663af57b error="update gateway state error: object does not exist"`

`object does not exist` error shows your gateway entity is not available in the database. I think you need to delete the gateway and add it to again in the Application Server.

closed 2025-05-02 by brocaar