brocaar/chirpstack-network-server #607: FCnt mismatch between GRPC API and WebInterface

vasalj, 2023-07-31:
- [ ] The issue is present in the latest release.
- [ ] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
I am trying to use gRPC API to send a downlink to the device. For the initial/first packet the down FCnt dosen't match the webUI FCnt.
```golang
devQueueServiceClient := asapi.NewDeviceQueueServiceClient(loraCommIface.grpcClient)
enqueueDevQueueItemReq := &asapi.EnqueueDeviceQueueItemRequest{DeviceQueueItem: devQueueItem}

if resp, err := devQueueServiceClient.Enqueue(context.Background(), enqueueDevQueueItemReq); err != nil {
        return
} else {
        fmt.Printf("Downlink message has been enqueued with FCnt: %d", resp.FCnt)
}
```
## What did you expect?
The FCnt that comes from gRPC is 0 but the FCnt that is really sent to the device is 1 (The acknowledgment of it also shows 1 rather than 0).

## Steps to reproduce this issue

Steps:

1.
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.17.6 |
| Network Server      | v3.16.2|
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2023-07-31:
Yes, this is possible. Because since the implementation of LoRaWAN 1.0.4 mac-commands get priority over application payloads (per specification). Therefore it might be that the downlink with fCnt=0 is used for mac-commands only and the application payload is re-enqueued with fCnt=1.

ChirpStack v4 solves this issue by giving you an unique identifier (UUID) on enqueue, which is also provided on ack or nack for correlation.

closed 2023-07-31 by brocaar

vasalj, 2023-07-31:
Thank you for the reply.
For seme reason, I cannot migrate to Chirpsatck V4 at the moment (I understand also you are already moved to V4 and you may not want to talk about the past!).

I just want to know if there is anyway I can get the real FCnt after re-enqueing is done by Chirpstack? (I send the downlink command with confirmation and use the FCnt to trace back the response/uplink for device)

vasalj, 2023-08-01:
[image]