brocaar/chirpstack-network-server #613: Misleading label for LoRa frames

vsaljooghi, 2024-05-08:
- [ ] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?
When I look in Chirpstack GUI for LoRaWAN Frames, the events naming/label is confusing.

For example: 
ConfirmedDataDown label uses a past tense (confirmed) to describe an event which will happen in future (confirmation/acknowledgment will happen with receive of next uplink)

## What did you expect?

ConfirmedDataUp --> DataUpRequiringConfirmation
ConfirmedDataDown --> DataDownRequiringConfirmation
UncomfirmedDataUp --> DataUpNotRequiringConfirmation
UncomfirmedDataDown --> DataDownNotRequiringConfirmation

## Steps to reproduce this issue

Steps:

1.
2.
3.
4.

## Could you share your log output?

```shell

```

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v?.?.?  |
| Network Server      |         |
| Gateway Bridge      |         |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2024-05-08:
The naming is aligned with the naming in the LoRaWAN specification (see
MType field in the specification). I'm not planning to use different MType
names that officially specified because this would become very confusing.

mmrein, 2024-05-09:
The proposed way is also roughtly 2x longer and far less readable. @vsaljooghi Look at it like this: ConfirmedDataUp = uplink data that must be **confirmed** (in the future)